stanford-ssi/samwise-flight-software #31: Replace LOG_ macros with logger abstraction.

devYaoYH, 2025-02-04:
Introduce a module `src/logger` to hide complexity of switching between TEST/RELEASE version of logging, also provides the interface to introduce functionality for FLASH/DISK logging behavior (when drivers are available).

One imagines a common interface like:

```c
/* Single main logging function
 * Parameters:
 *  LOG_LEVEL: DEBUG/INFO/ERROR
 *  SINK_BITMASK: 0bxyz (x - FLASH, y - DISK, z - TEST)
 *  fmt: printf-like format string
 *  args: variable args
 */
void log(enum LOG_LEVEL, uint8 SINK_BITMASK, char* fmt, args...)

// Macros for convenience & backward compatibility with current code
// Notice the `| TEST` bitwise operator here to automatically enable TESTING mode based on
// define flag passed through from cmake build commands.
LOG_DEBUG(fmt, ...)                                                    \
    log(DEBUG, 0b010 | TEST, "[DEBUG]    " fmt "\n" __VA_OPT__(, ) __VA_ARGS__)
```

Darrow8, 2025-02-15:
I have created `logger` branch for this, work in progress right now.

devYaoYH, 2025-05-24:
Thanks @Darrow8 & @shetaye for working on this! This should be completed now, `src/drivers/logger` interface exists :)