stanford-ssi/samwise-flight-software #63: Satellite 433 Radio Configuration

devYaoYH, 2025-05-13:
Frequency: 438.1 MHz
Coding Rate: 5
Spreading Factor: 7
Baudrate: 125 KHz

CRC (Cyclic Redundancy Check): ON
LDRO (Low Data Rate Optimization): ON
AGC (Automatic Gain Control): ON

devYaoYH, 2025-05-13:
Some settings to note for LoRA radio:

[image]

devYaoYH, 2025-05-13:
Instructions for SSH into Ground Station & Flat Sat setup: https://docs.google.com/document/d/1cDkb7PT_642zS3IgYS5149CFJi5kesSO_I0K0GwfCvY/edit?tab=t.0

devYaoYH, 2025-05-13:
Things we want to check during Range Test:
1. Tx power histogram (at what power should our satellite be transmitting on)
2. Link rate (data rate for decoding messages from ground station)
3. 2.4 GHz downlink test

devYaoYH, 2025-05-24:
See additional instructions on 1. power histogram in #45

megargayu, 2026-05-04:
@devYaoYH @lc0001coll wanted to check - has this issue been resolved/finished?