stanford-ssi/samwise-flight-software #74: [Bug] Potential race condition for LoRA transmission queue

devYaoYH, 2025-05-23:
From PR #70 review:

> Also consider reverting this change, I think it's a no-op if the radio task runs and there are no packets to be transmitted in the tx_queue (so we can run it more frequently anyhow to catch pending packets added by other tasks).
>
>But... I'm cognizant of the race-condition when we execute this and a packet transmission is already underway and we club the single FIFO queue on the rfm9x with another call to tx_done(). Any chance we could fix this with a simple 'locking' mechanism that writes a value into slate whenever we first enter tx_done and writes another value once tx_queue is empty and it switches back into receive mode (else branch starting line 34).
>
>This is once again out-of-scope for your current PR, so consider rolling back this change for now?
>
>QQ: do you know if queue_t in pico is multithread-safe? (atomic pop/push semantics)

shetaye, 2025-05-25:
re QQ: yes queue_t is thread safe (https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#group_queue)

devYaoYH, 2025-07-13:
One potential solution is to add a check for whether TX is underway when we attempt to enter TX mode again from the radio_task main loop.

devYaoYH, 2025-09-24:
Probably going to be fixed as of PR #140