stanford-ssi/samwise-flight-software #127: [ADCS] Integrate UART comms protocol with current PiCubed software

devYaoYH, 2025-07-13:
From @niklas-vainio  Not sure if I sent this already, but before I forget here’s the main task for the picubed to interface with the ADCS board
telem_picubed_side.cpp

```c
/**
 * @author Niklas Vainio
 * @date 2025-05-27
 *
 * This code defines the picuebd side of the ADCS uart interface
 * telem_picubed_side.cpp
 */
​
#include "hardware/irq.h"
#include "hardware/uart.h"
#include "pico/stdlib.h"
#include "pico/util/queue.h"
#include "slate.h"
​
#define ADCS_TELEM_QUEUE_SIZE (4 * sizeof(adcs_packet_t))
​
static slate_t *slate_for_irq;
static bool escape_active = false;
static bool packet_active = false;
static uint32_t current_packet_bytes = 0;
​
/**
 * Simple helper function to clear a queue
 */
static void queue_clear(queue_t *q)
{
    char c;
    while (queue_try_remove(q, &c))
    {
    };
}
​
static void reset_packet_state()
{
    packet_active = false;
    escape_active = false;
    current_packet_bytes = 0;
    queue_clear(&slate_for_irq->adcs_telemetry_queue);
​
    slate_for_irq->adcs_telem_valid = false;
}
​
/**
 * @brief Helper function to process a byte coming in over uart
 *
 * @param byte
 */
static void handle_uart_byte(char byte)
{
    // No packet active: activate if start otherwise do nothing
    if (!packet_active)
    {
        if (byte == SLIP_START)
        {
            reset_packet_state();
            packet_active = true;
        }
​
        return;
    }
​
    // Packet is active: if this is a start byte, this signifies the end of the
    // packet
    if (byte == SLIP_START)
    {
        // Verify length of current packet - reset if not correct
        if (current_packet_bytes == sizeof(adcs_packet_t))
        {
            // Copy to slate
            for (int i = 0; i < current_packet_bytes; i++)
            {
                char c;
                if (queue_try_remove(&slate_for_irq->adcs_telemetry_queue, &c))
                {
                    ((char *)&slate_for_irq->adcs_telem)[i] = c;
                }
            }
​
            // Start new packet (stay active)
            current_packet_bytes = 0;
            slate_for_irq->adcs_telem_valid = true;
        }
        else
        {
            reset_packet_state();
        }
​
        return;
    }
​
    // Escape byte: set escape active flag
    if (byte == SLIP_ESC)
    {
        escape_active = true;
        return;
    }
​
    // Handle possible escape
    char byte_to_add = byte;
​
    if (escape_active)
    {
        if (byte == SLIP_ESC_ESC)
        {
            // Escaped escape
            byte_to_add = SLIP_ESC;
            escape_active = false;
        }
        else if (byte == SLIP_ESC_START)
        {
            // Escaped start
            byte_to_add = SLIP_START;
            escape_active = false;
        }
        else
        {
            // Other: reset packet
            reset_packet_state();
            return;
        }
    }
​
    // Otherwise, this is a normal byte - just add it to the queue
    bool result = queue_try_add(&slate_for_irq->adcs_telemetry_queue, &byte);
    current_packet_bytes++;
​
    if (!result)
    {
        // No space in queue: invalidate this packet
        reset_packet_state();
        return;
    }
}
​
/**
 * @brief IRQ function called when we receive a uart byte
 */
static void on_byte_rx()
{
    // Read all bytes
    while (uart_is_readable(SAMWISE_ADCS_UART))
    {
        uint8_t ch = uart_getc(SAMWISE_ADCS_UART);
        handle_uart_byte(ch);
    }
}
​
/**
 * Initialize interrupt handler for receiving from ADCS hardware
 */
void adcs_telemetry_task_init(slate_t *slate)
{
    // Initialize receiving queue
    queue_init(&slate->adcs_telemetry_queue, 1, ADCS_TELEM_QUEUE_SIZE);
​
    // Initialize the uart for receiving with an interrupt handler
    uart_init(SAMWISE_ADCS_UART, PICUBED_UART_BAUD);
    gpio_set_function(
        SAMWISE_ADCS_TX_TO_PICUBED,
        UART_FUNCSEL_NUM(SAMWISE_ADCS_UART, SAMWISE_ADCS_TX_TO_PICUBED));
    gpio_set_function(
        SAMWISE_ADCS_RX_FROM_PICUBED,
        UART_FUNCSEL_NUM(SAMWISE_ADCS_UART, SAMWISE_ADCS_RX_FROM_PICUBED));
​
    // Set data format
    uart_set_format(SAMWISE_ADCS_UART, PICUBED_UART_DATA_BITS,
                    PICUBED_UART_STOP_BITS, PICUBED_UART_PARITY);
​
    // Turn off FIFOs
    uart_set_fifo_enabled(SAMWISE_ADCS_UART, false);
​
    // Set up IRQ
    int UART_IRQ = SAMWISE_ADCS_UART == uart0 ? UART0_IRQ : UART1_IRQ;
​
    // And set up and enable the interrupt handlers
    irq_set_exclusive_handler(UART_IRQ, on_byte_rx);
    irq_set_enabled(UART_IRQ, true);
​
    uart_set_irq_enables(SAMWISE_ADCS_UART, true, false);
​
    // Bookmark slate and reset state
    slate_for_irq = slate;
    reset_packet_state();
}
​
void adcs_telemetry_task_dispatch(slate_t *slate)
{
    // (Do not need to do anything, interrupts take care of everything)
}
```

you might also need this struct that defines the telemetry packet - this will change before launch though, I will send you guys an updated version (and please ping me if I forget!)

```c
/**
 * @author Niklas Vainio
 * @date 2025-05-27
 *
 * This file defines the ADCS telemetry struct.
 * IMPORTANT: KEEP UP TO DATE WITH THE PICUBED
 *
 * Last updated: 05/27/2025
 *
 * (this must live in a separate file for C include reasons)
 * adcs_packet.h
 */
​
#pragma once
​
#include <stdint.h>
​
typedef struct __attribute__((packed))
{
    // Angular velocity
    float w;
​
    // Quaternion Estimate
    float q0, q1, q2, q3;
​
    // Misc Data
    char state;
    uint32_t boot_count;
​
} adcs_packet_t;
```

Currently these are standalone scripts and we need to integrate it with our software.

devYaoYH, 2025-09-24:
@lundeen06 has this been tested yet?

lundeen06, 2025-09-24:
No, this version of Picubed <-> ADCS UART has not been tested. 

Related, how urgent is it for me to populate our adcs telemetry? Been procrastinating it but can get it done asap, just let me know the relative urgency. Potentially could make software frosh & gnc frosh do it, and both review their PRs. Or we bang it out, lmk.